linux-nvme/libnvme #613: libnvme using deprecated json-c APIs

cleech, 2023-04-04:
This popped up in some testing with a libnvme built against json-c 0.14 that was then installed on a system with json-c 0.15.  The printbuf APIs were moved from the JSONC_0.14 tag to JSONC_PRIVATE, breaking the json-c ABI.  They continue to work if recompiling against 0.15+, but are deprecated for external use and may be removed in a future version.

It looks like the printbuf is being used as a re-allocating heap buffer during JSON parsing only, and it's use should be removed.

cleech, 2023-04-04:
I think we can pass the read buffers to json_tokener_parse_ex incrementally, and not need a flexible buffer to hold a large file at all.  Or maybe just mmap the file.

I can work up merge request for this.

ikegami-t, 2023-06-20:
The PR #664 merged into master so resolved.

closed 2023-06-21 by igaw