linux-nvme/libnvme #616: Question about the judgement( offset & bs ) in function nvme_bytes_to_lba

Rookie-roob, 2023-04-05:
I have a doubt in the judgement in function **nvme_bytes_to_lba** .
Relevant code is shown below .

`
if (!count || offset & bs || count & (bs - 1))
`

I have a doubt in the judgement ("offset & bs")  and I wonder if " offset & bs" should be modified to "offset & (bs - 1)"

Thanks for your read.

keithbusch, 2023-04-05:
Yes, you are correct.

igaw, 2023-04-06:
I'll prepare a fix... Man I could have spotted that too :D

sunyb-dot, 2023-04-08:
I also have a question about this function, why are we substracting 1 here?
```
*nlb = (count >> n->lba_shift) - 1;
```

keithbusch, 2023-04-08:
The subtraction is correct because this value is 'zeroes based". IOW, zero means one. Just an nvme idiosyncracy.

PR #620 (2023-04-12): tree: Fix offset argument check in nvme_bytes_to_lba
Also offset modulo blocksize needs to be 0. Commit 01c6055e5602 ("tree: Fix argument check in nvme_bytes_to_lba") missed to update this, thus do it now.

Fixes #616
closed 2023-04-12 by igaw