linux-nvme/libnvme #619: Refactor use of nvme_msg in nbft.c

johnmeneghini, 2023-04-12:
```
Functions calling this function: nvme_msg

  File   Function          Line
0 nbft.c verify             84 nvme_msg(NULL, LOG_DEBUG, "file %s: " message "\n", \
1 nbft.c __get_heap_obj     99 nvme_msg(NULL, LOG_DEBUG,
2 nbft.c __get_heap_obj    110 nvme_msg(NULL, LOG_DEBUG,
3 nbft.c __get_heap_obj    117 nvme_msg(NULL, LOG_DEBUG,
4 nbft.c read_ssns         231 nvme_msg(NULL, LOG_DEBUG,
5 nbft.c read_ssns         264 nvme_msg(NULL, LOG_DEBUG,
6 nbft.c read_ssns         282 nvme_msg(NULL, LOG_DEBUG,
7 nbft.c read_ssns         291 nvme_msg(NULL, LOG_DEBUG,
8 nbft.c read_hfi_info_tcp 335 nvme_msg(NULL, LOG_DEBUG,
9 nbft.c read_hfi          412 nvme_msg(NULL, LOG_DEBUG,
a nbft.c read_discovery    454 nvme_msg(NULL, LOG_DEBUG,
b nbft.c read_discovery    460 nvme_msg(NULL, LOG_DEBUG,
c nbft.c nvme_nbft_read    695 nvme_msg(NULL, LOG_ERR, "Failed to open %s: %s\n",
d nbft.c nvme_nbft_read    703 nvme_msg(NULL, LOG_ERR, "Failed to read from %s: %s\n",
e nbft.c nvme_nbft_read    715 nvme_msg(NULL, LOG_ERR, "Failed to allocate memory for NBFT table");
f nbft.c nvme_nbft_read    723 nvme_msg(NULL, LOG_ERR, "Failed to read from %s: %s\n",
g nbft.c nvme_nbft_read    737 nvme_msg(NULL, LOG_ERR, "Could not allocate memory for NBFT\n");
h nbft.c nvme_nbft_read    748 nvme_msg(NULL, LOG_ERR, "Failed to parse %s\n", filename);
```

johnmeneghini, 2023-04-12:
See the discussion in: https://github.com/linux-nvme/libnvme/pull/572#discussion_r1158343257

igaw, 2023-09-20:
So with the merge of #669 we have a global root object (sic) which can be used for logging. This global root object is used used when `nvme_msg` is called with `NULL`. So this should behave slightly better.

closed 2023-09-20 by igaw