linux-nvme/libnvme #634: swig warning during libnvme build

martin-gpy, 2023-04-24:
Seeing the below warning during libnvme build on a SLES15 SP4 MU host:

```
# ninja -C .build
ninja: Entering directory `.build'
[46/59] Generating nvme.py with a custom command
notype name zns
../src/nvme/types.h:7727: Warning 503: Can't wrap class $unnamed7$ unless renamed to a valid identifier.
[59/59] Linking target libnvme/_nvme.cpython-36m-x86_64-linux-gnu.so
```
```
# rpm -qa | grep gcc
gcc-7-3.9.1.x86_64
gcc-c++-7-3.9.1.x86_64
libgcc_s1-11.2.1+git610-150000.1.6.6.x86_64
libstdc++6-pp-gcc11-11.2.1+git610-150000.1.6.6.x86_64
libstdc++6-devel-gcc7-7.5.0+r278197-4.30.1.x86_64
gcc7-7.5.0+r278197-4.30.1.x86_64
gcc7-c++-7.5.0+r278197-4.30.1.x86_64
```

martin-belanger, 2023-04-24:
Yep. Saw that too. Seems to be a bug with swig. We've configured swig to ignore structs, classes, unions, etc., but it still complains about anonymous structs, classes, unions, etc. This is reference to an anonymous union here: https://github.com/linux-nvme/libnvme/blob/master/src/nvme/types.h#L7725

@igaw - would it be OK if we added a name to this union? We may have to add a name to the struct inside as well.

martin-belanger, 2023-04-25:
Submitted https://github.com/linux-nvme/libnvme/pull/636

martin-gpy, 2023-04-27:
@martin-belanger, the swig 503 warning no longer shows up now with your change, but I still see 'notype name zns' displayed during the build:

```
# ninja -C .build
ninja: Entering directory `.build'
[47/59] Generating nvme.py with a custom command
notype name zns
[59/59] Linking target libnvme/_nvme.cpython-36m-x86_64-linux-gnu.so
```

tbzatek, 2023-04-27:
I guess it really depends on swig version: `swig-4.1.1-4.fc38.x86_64`

```
[57/853] Generating libnvme/nvme.py with a custom command
Deprecated command line option: -py3. Ignored, this option is no longer supported.
notype name zns
../src/nvme/types.h:7727: Warning 503: Can't wrap class $unnamed7$ unless renamed to a valid identifier.
```

igaw, 2023-04-27:
The compiler is not really happy with just giving the struct a name:
```shell
../src/nvme/types.h:7727:24: error: ‘struct nvme_ns_mgmt_host_sw_specified::<unnamed union>::zns’ invalid; an anonymous union may only have public non-static data members [-fpermissive]
 7727 |                 struct zns {
      |  
```

So something like this would be needed:
```diff
@@ -7722,15 +7722,15 @@ struct nvme_ns_mgmt_host_sw_specified {
        __le64                  lbstm;
        __le16                  nphndls;
        __u8                    rsvd394[105];
-       union {
+       union foo {
                __u8            rsvd499[13];
-               struct {
+               struct zns {
                        __u8    znsco;
                        __le32  rar;
                        __le32  ror;
                        __le32  rnumzrwa;
                } __attribute__((packed)) zns;
-       };
+       } foo;
        __le16                  phndl[128];
        __u8                    rsvd768[3328];
 };
```

martin-belanger, 2023-04-27:
Check this PR: https://github.com/linux-nvme/libnvme/pull/638

This solves the issue with SWIG and we don't have to modify the struct/union.

igaw, 2023-04-27:
Well, even if I don't like the idea but in this case we just declare the zns data struct but clearly state in the documentation it should not be used anywhere (see also https://github.com/linux-nvme/libnvme/pull/582)

martin-belanger, 2023-04-27:
> But this would also make the data struct unavailable in Python, right? I fear this is not going to be a long time solution :)

Right, but we don't want to make the struct available in Python. We only care about the `enums` and `#defines`. In fact, in [nvme.i we specifically tell swig to ignore all struct](https://github.com/linux-nvme/libnvme/blob/master/libnvme/nvme.i#L1057).

martin-belanger, 2023-04-27:
I think it's a bug with swig. We specifically tell it to ignore structs (i.e. to not generate wrappers for the structs). However, swig still parses the structs and spits out warnings when something is not quite right.

igaw, 2023-04-27:
Ah, didn't realize this. In this case we get away with the ifdefery.

igaw, 2023-04-27:
Thanks @martin-belanger, I've merged the fix.

closed 2023-04-27 by igaw