linux-nvme/libnvme #672: Possible file descriptor leak (i.e. many namespace devices are left opened)

martin-belanger, 2023-06-29:
In nvme-stas, I use the Python bindings to create connections to controllers. A daemon creates a "root" and a "host" objects. Then a controller object is created and the "connect" method is invoked for that object (this invokes the C function `nvmf_add_ctrl`). These 3 objects (root, host, ctrl) remain in existence for as long as the daemon is running (and as long as nobody manually tells the daemon to remove the connection to the controller).

Using `lsof -p [daeemon-pid]` (see output below), I observe that the daemon has a lot of namespace devices opened. I'm just wondering why. This doesn't scale. As the daemon connects to more controllers the number of opened namespace devices keeps increasing until we hit the daemon's limit of opened files (default 1024).

I know that libnvme scans the sysfs to build a picture of what nvme devices are already in the system. Is it possible that we're leaking file descriptors (i.e. we forget to call `close()`) while scanning the sysfs. I looked at the code, but can't find where this is happening.

```
# lsof -p 42683
COMMAND   PID USER   FD      TYPE             DEVICE SIZE/OFF   NODE NAME
python3 42683 root  cwd       DIR               0,49      272    256 /
python3 42683 root  rtd       DIR               0,49      272    256 /
python3 42683 root  txt       REG               0,49    10560 195219 /usr/bin/python3.6
python3 42683 root  mem       REG               0,49    15376 134253 /usr/lib64/python3.6/site-packages/systemd/id128.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    41640 134251 /usr/lib64/python3.6/site-packages/systemd/_reader.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    10896 134250 /usr/lib64/python3.6/site-packages/systemd/_journal.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    97112 194704 /usr/lib64/python3.6/lib-dynload/_pickle.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49   219592 183508 /usr/lib64/libudev.so.1.7.2
python3 42683 root  mem       REG               0,49    84120 194692 /usr/lib64/python3.6/lib-dynload/_datetime.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49   122928 194691 /usr/lib64/python3.6/lib-dynload/_ctypes.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    18792   9144 /usr/lib64/libkeyutils.so.1.10
python3 42683 root  mem       REG               0,49    64208   9149 /usr/lib64/libjson-c.so.3.0.1
python3 42683 root  mem       REG               0,49   163656 302581 /usr/lib64/libnvme.so.1.4.0
python3 42683 root  mem       REG               0,49    25704 194716 /usr/lib64/python3.6/lib-dynload/binascii.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    18520 194718 /usr/lib64/python3.6/lib-dynload/fcntl.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49   181664 302588 /usr/lib64/python3.6/site-packages/libnvme/_nvme.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    80488 194709 /usr/lib64/python3.6/lib-dynload/_sha3.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    26600 183606 /usr/lib64/libjitterentropy.so.3.4.0
python3 42683 root  mem       REG               0,49  3389800 292194 /usr/lib64/libcrypto.so.1.1
python3 42683 root  mem       REG               0,49   146592   9187 /usr/lib64/libgpg-error.so.0.32.0
python3 42683 root  mem       REG               0,49  1296440 287278 /usr/lib64/libgcrypt.so.20.3.4
python3 42683 root  mem       REG               0,49    39680   9387 /usr/lib64/libcap.so.2.63
python3 42683 root  mem       REG               0,49   133544   9127 /usr/lib64/liblz4.so.1.9.3
python3 42683 root  mem       REG               0,49  1244088 183476 /usr/lib64/libzstd.so.1.5.0
python3 42683 root  mem       REG               0,49    44904 179898 /lib64/librt-2.31.so
python3 42683 root  mem       REG               0,49   922648 187535 /usr/lib64/libsystemd.so.0.32.0
python3 42683 root  mem       REG               0,49    10808 194728 /usr/lib64/python3.6/lib-dynload/syslog.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    15496 194706 /usr/lib64/python3.6/lib-dynload/_random.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    42256 194681 /usr/lib64/python3.6/lib-dynload/_blake2.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49    21176 134249 /usr/lib64/python3.6/site-packages/systemd/_daemon.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49   359716  52574 /usr/lib64/girepository-1.0/Gio-2.0.typelib
python3 42683 root  mem       REG               0,49    91784 194711 /usr/lib64/python3.6/lib-dynload/_socket.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49   205596  52571 /usr/lib64/girepository-1.0/GLib-2.0.typelib
python3 42683 root  mem       REG               0,49   211696  28488 /usr/lib64/python3.6/site-packages/cairo/_cairo.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49   133104 241236 /usr/lib64/libGLX.so.0.0.0
python3 42683 root  mem       REG               0,49    14552   9550 /usr/lib64/libXau.so.6.0.0
python3 42683 root  mem       REG               0,49   710792 241238 /usr/lib64/libGLdispatch.so.0.0.0
python3 42683 root  mem       REG               0,49    30848 287424 /usr/lib64/libuuid.so.1.3.0
python3 42683 root  mem       REG               0,49   563256 241230 /usr/lib64/libGL.so.1.7.0
python3 42683 root  mem       REG               0,49    73440  17961 /usr/lib64/libXext.so.6.4.0
python3 42683 root  mem       REG               0,49  1314384 184065 /usr/lib64/libX11.so.6.3.0
python3 42683 root  mem       REG               0,49    43576  17949 /usr/lib64/libXrender.so.1.3.0
python3 42683 root  mem       REG               0,49    55648 138063 /usr/lib64/libxcb-render.so.0.0.0
python3 42683 root  mem       REG               0,49   166872 137216 /usr/lib64/libxcb.so.1.1.0
python3 42683 root  mem       REG               0,49    14480 138051 /usr/lib64/libxcb-shm.so.0.0.0
python3 42683 root  mem       REG               0,49   272368   9764 /usr/lib64/libpng16.so.16.34.0
python3 42683 root  mem       REG               0,49    79896 241228 /usr/lib64/libEGL.so.1.1.0
python3 42683 root  mem       REG               0,49   777520 138005 /usr/lib64/libfreetype.so.6.17.4
python3 42683 root  mem       REG               0,49   281672  12014 /usr/lib64/libfontconfig.so.1.12.0
python3 42683 root  mem       REG               0,49   691024 183560 /usr/lib64/libpixman-1.so.0.40.0
python3 42683 root  mem       REG               0,49    34952  18908 /usr/lib64/libcairo-gobject.so.2.11600.0
python3 42683 root  mem       REG               0,49  1266136  18597 /usr/lib64/libcairo.so.2.11600.0
python3 42683 root  mem       REG               0,49    18912 200123 /usr/lib64/python3.6/site-packages/gi/_gi_cairo.cpython-36m-x86_64-linux-gnu.so
python3 42683 root  mem       REG               0,49   343208 287267 /usr/lib64/libblkid.so.1.1.0
python3 42683 root  mem       REG               0,49   101080 179896 /lib64/libresolv-2.31.so
python3 42683 root  mem       REG               0,49   159704   8590 /lib64/libselinux.so.1
python3 42683 root  mem       REG               0,49   387880 289353 /usr/lib64/libmount.so.1.1.0
python3 42683 root  mem       REG               0,49  1961648 240170 /usr/lib64/libgio-2.0.so.0.7000.5
python3 42683 root  mem       REG               0,49   559440 137308 /usr/lib64/libpcre.so.1.2.13
python3 42683 root  mem       REG               0,49    39440   9285 /usr/lib64/libffi.so.7.1.0
python3 42683 root  mem       REG               0,49   221272  52562 /usr/lib64/libgirepository-1.0.so.1.0.0
[... 181 lines cut ...]
python3 42683 root  136r      BLK             259,11      0t0   1017 /dev/nvme5n3
python3 42683 root  137r      BLK             259,15      0t0   1023 /dev/nvme5n4
python3 42683 root  138r      BLK              259,9      0t0   1012 /dev/nvme6n1
python3 42683 root  139r      BLK             259,19      0t0   1041 /dev/nvme6n2
python3 42683 root  140r      BLK             259,23      0t0   1057 /dev/nvme6n3
python3 42683 root  141r      BLK             259,29      0t0   1070 /dev/nvme6n4
[... cut ...]
python3 42683 root  152u  a_inode               0,14        0  10594 [eventfd]
python3 42683 root  153u     unix 0xffff8f218b6baec0      0t0 780180 type=STREAM
python3 42683 root  154u     unix 0xffff8f218b6bbfc0      0t0 780181 type=STREAM
python3 42683 root  155r      CHR              243,0      0t0    971 /dev/nvme0
python3 42683 root  156r      CHR              243,3      0t0    980 /dev/nvme3
python3 42683 root  157r      CHR              243,1      0t0   2468 /dev/nvme1
python3 42683 root  158r      CHR              243,2      0t0   2471 /dev/nvme2

```

martin-belanger, 2023-06-29:
It took a while, but I finally got some info with Valgrind. Here's one (of many) examples.
```
==59285== Open file descriptor 12: /dev/nvme16n3
==59285==    at 0x4B52E6D: open (in /lib64/libpthread-2.31.so)
==59285==    by 0xBA98F48: nvme_open (in /usr/lib64/libnvme.so.1.4.0)
==59285==    by 0xBA9C8FF: ??? (in /usr/lib64/libnvme.so.1.4.0)
==59285==    by 0xBA9CAC9: ??? (in /usr/lib64/libnvme.so.1.4.0)
==59285==    by 0xBA9DA5D: nvme_scan_topology (in /usr/lib64/libnvme.so.1.4.0)
==59285==    by 0xBA9DC83: nvme_scan (in /usr/lib64/libnvme.so.1.4.0)
==59285==    by 0xBA61604: ??? (in /usr/lib64/python3.6/site-packages/libnvme/_nvme.cpython-36m-x86_64-linux-gnu.so)
==59285==    by 0x491608C: _PyCFunction_FastCallDict (in /usr/lib64/libpython3.6m.so.1.0)
==59285==    by 0x4987A71: ??? (in /usr/lib64/libpython3.6m.so.1.0)
==59285==    by 0x498BED4: _PyEval_EvalFrameDefault (in /usr/lib64/libpython3.6m.so.1.0)
==59285==    by 0x49874D5: ??? (in /usr/lib64/libpython3.6m.so.1.0)
==59285==    by 0x498F798: _PyFunction_FastCallDict (in /usr/lib64/libpython3.6m.so.1.0)
```

So it is indeed the `nvme_scan_topology()` that leaves all the FDs open. 

In this particular case, `stafd` (the daemon) connects to several Discovery Controllers. I guess `nvme_scan_topology` doesn't care whether we're only interested by discovery controllers and scans everything including namespace devices.

igaw, 2023-06-30:
`nvme_scan_topology()` is written in the way that it is iterating over the sysfs tree and only adds entries to its internal tree. This wasn't an issue so far, because nvme-cli always using `nvme_scan_topololy()` on an empty internal tree. 

I understand `nvme-stas` is keeping the `nvme_root_t` object open all the time and thus it will eat up all resources eventually. 

That means we need to extend `nvme_scan_topology()` to also remove nodes from the internal tree when iterating over sysfs. Kind of obvious now :)

igaw, 2023-06-30:
So the real fix is to get rid of the fiel descriptor completely. Currently it is used to issue id commands which in theory shouldn't be necessary, all necessary info should be in sysfs.

martin-belanger, 2023-06-30:
I agree. And for the few cases where we need to send an `ioctl()` commands, we can simply momentarily `open()`/`write()`/`read()`/`close()` the file. No need to keep the file opened forever. Especially the block device files, which we never actually do anything with.

martin-belanger, 2023-06-30:
@igaw - I'll be working on a patch.

I don't see `nvme_ctrl_get_fd()` or `nvme_ns_get_fd()` being used outside of `libnvme`. Would it be OK to remove these two functions from the public API and only use them internally in `libnvme`. This way we can  momentarily `open()` the device when needed and invoke `close()` after we're done with it?

igaw, 2023-07-03:
We can try but if someone complains we have to add it back.

martin-belanger, 2023-07-03:
> Or we just add an `open()` into `nvme_[ctrl|ns]_get_fd()` directly. If the caller is not using these function no `fd` leak.

Yep! I changed my mind about removing `nvme_[ctrl|ns]_get_fd()` because a lot of the public APIs require `fd` as an argument. Sure, people could instead use `nvme_open(nvme_[ctrl|ns]_get_name(c))` to get the `fd`, but that gets complicated. Instead, I will do like you said (i.e add an `open()` into `nvme_[ctrl|ns]_get_fd()`) and document that the caller is responsible for closing that `fd` when they are done with it.

igaw, 2023-07-03:
Just an idea, we could still keep track of the `fd` inside libnvme but we only open it lazy.

closed 2023-07-20 by igaw