linux-nvme/libnvme #682: nvme list fw rev does not get updated after firmware update

tkenbo, 2023-07-20:
`
[root@host ~]# nvme version
nvme version 2.5 (git 2.5)
libnvme version 1.5 (git 1.5-9-g1b042eb)
[root@host ~]# nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme2n1          /dev/ng2n1            8WA00008             XP0000000000                            0x1          6.40  TB /   6.40  TB    512   B +  0 B   AAAAA470
[root@host ~]# nvme fw-download /dev/nvme2 -f AAAAA480.bin
Firmware download success
[root@host ~]# nvme fw-commit /dev/nvme2 -s 2 -a 1
Success committing firmware action:1 slot:2
[root@host ~]# nvme reset /dev/nvme2
[root@host ~]# nvme list
Node                  Generic               SN                   Model                                    Namespace  Usage                      Format           FW Rev
--------------------- --------------------- -------------------- ---------------------------------------- ---------- -------------------------- ---------------- --------
/dev/nvme2n1          /dev/ng2n1            8WA00008             XP00000000000                            0x1          6.40  TB /   6.40  TB    512   B +  0 B   AAAAA470
[root@host ~]# nvme id-ctrl /dev/nvme2 | grep fr
fr        : AAAAA480
[root@host ~]# cat /sys/class/nvme/nvme2/firmware_rev
AAAAA470
`

This used work as expected with v1.x as scan_ctrl function called identify controller command.
`ret = nvme_identify_ctrl(fd, &c->id);`

However in v2.x, it has changed to populate it from sysfs information.
`c->firmware = nvme_get_ctrl_attr(c, "firmware_rev");`

As shown above, firmware_rev attribute in sysfs does not seem changing after firmware update and not sure if this is how is it expected, incorrect driver behavior or nvmecli issue.
It'd be much appreciated if anyone could advise.

igaw, 2023-07-21:
Duplicate of #1991

keithbusch, 2023-09-27:
What firmware commit action did you use when this issue was seen?

tkenbo, 2023-09-27:
Yes.  I did reset.  id-ctrl reports back the correct value but sysfs did not.

```
[root@host ~]# nvme fw-download /dev/nvme2 -f AAAAA480.bin
Firmware download success
[root@host ~]# nvme fw-commit /dev/nvme2 -s 2 -a 1
Success committing firmware action:1 slot:2
[root@host ~]# nvme reset /dev/nvme2
[root@host ~]# nvme id-ctrl /dev/nvme2 | grep fr
fr        : AAAAA480
[root@host ~]# cat /sys/class/nvme/nvme2/firmware_rev
AAAAA470
```

igaw, 2023-09-29:
@keithbusch: What about this idea here?

```patch
From 852bf0fa04990bae112ae2d9572c245db5e8d9e0 Mon Sep 17 00:00:00 2001
From: Daniel Wagner <dwagner@suse.de>
Date: Fri, 29 Sep 2023 14:55:49 +0200
Subject: [PATCH] nvme: update firmware revision after firmware commit

After successfully committing a new firmware update the firmware
revision information.

---
 drivers/nvme/host/core.c | 19 +++++++++++++++++++
 1 file changed, 19 insertions(+)

diff --git a/drivers/nvme/host/core.c b/drivers/nvme/host/core.c
index 37b6fa746662..324426986a1e 100644
--- a/drivers/nvme/host/core.c
+++ b/drivers/nvme/host/core.c
@@ -125,6 +125,7 @@ static void nvme_remove_invalid_namespaces(struct nvme_ctrl *ctrl,
 					   unsigned nsid);
 static void nvme_update_keep_alive(struct nvme_ctrl *ctrl,
 				   struct nvme_command *cmd);
+static int nvme_identify_ctrl(struct nvme_ctrl *dev, struct nvme_id_ctrl **id);
 
 void nvme_queue_scan(struct nvme_ctrl *ctrl)
 {
@@ -1124,6 +1125,20 @@ u32 nvme_passthru_start(struct nvme_ctrl *ctrl, struct nvme_ns *ns, u8 opcode)
 }
 EXPORT_SYMBOL_NS_GPL(nvme_passthru_start, NVME_TARGET_PASSTHRU);
 
+static void nvme_update_firmware_rev(struct nvme_ctrl *ctrl)
+{
+	struct nvme_id_ctrl *id;
+	int ret;
+
+	ret = nvme_identify_ctrl(ctrl, &id);
+	if (ret) {
+		dev_err(ctrl->device, "Identify Controller failed (%d)\n", ret);
+		return;
+	}
+	memcpy(ctrl->subsys->firmware_rev, id->fr,
+	       sizeof(ctrl->subsys->firmware_rev));
+}
+
 void nvme_passthru_end(struct nvme_ctrl *ctrl, struct nvme_ns *ns, u32 effects,
 		       struct nvme_command *cmd, int status)
 {
@@ -1163,6 +1178,10 @@ void nvme_passthru_end(struct nvme_ctrl *ctrl, struct nvme_ns *ns, u32 effects,
 			break;
 		}
 		break;
+	case nvme_admin_activate_fw:
+		if (!status)
+			nvme_update_firmware_rev(ctrl);
+		break;
 	default:
 		break;
 	}
-- 
2.42.0
```

igaw, 2023-10-06:
@tkenbo is there any chance you could test the above patch?

tkenbo, 2023-10-09:
@igaw - I don't think I've ever built a native driver before.  Do you know if there is any reference I can read on?

igaw, 2023-10-09:
It's not too complicated. The first step is to install all build dependencies. Most distribution have some sort of template/pattern for this. E.g. 

```shell
# apt-get install build-essential
# apt-get build-dep linux
```
https://kernel-team.pages.debian.net/kernel-handbook/ch-common-tasks.html#s-common-official

After you got all the tools installed you can follow the kernel guide on how to build your kernel.

https://docs.kernel.org/admin-guide/quickly-build-trimmed-linux.html#step-by-step-guide

Usually, I don't want to figure out which module I need and just use the distribution configuration as base config. So this is what I typically do:

```shell
$ git clone https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git
$ cd linux
[ apply patch ]
$ cp /boot/config-6.5.3-1-default .config
$ make olddefconfig
$ make -j $(nproc)
$ sudo make modules_install install
```

And then reboot and select the new kernel. After testing you need to remove the kernel by hand as with this method it's not registered with the package manager of your distribution. This is also fairly simply to do:

```shell
# rm -rf /lib/modules/${KERNEL_VERSION}
# rm /boot/*${KERNEL_VERSION}*
# grub2-mkconfig > /boot/grub2/grub.cfg
```

Obviously double check the above commands match with your installation before executing. This could break your system.

igaw, 2023-10-11:
After looking at the kernel code again, I think we  might want add this part to nvme_fw_act_work() instead. As I understand the spec, the firmware is supposed to send AEN after we commit the new firmware. So this would be the best place to update the firmware version I suppose.

tkenbo, 2023-10-12:
finally I was able to compile the kernel with the change.  I put the change in nvme_fw_act_work() and it worked as expected.

```
static void nvme_fw_act_work(struct work_struct *work)
{
        .....
        nvme_unquiesce_io_queues(ctrl);
        /* read FW slot information to clear the AER */
        nvme_get_fw_slot_info(ctrl);
        nvme_update_firmware_rev(ctrl);

        queue_work(nvme_wq, &ctrl->async_event_work);
}
```

igaw, 2023-10-12:
Excellent! I'll prepare a kernel patch. Thanks for testing. Really appreciated!

igaw, 2023-10-13:
https://lore.kernel.org/linux-nvme/20231013062623.6745-1-dwagner@suse.de/

@tkenbo if you wish, I'll add you as reporter and tester. If so, I'd need your either your email address, or you just reply to the email from above yourself.

igaw, 2023-11-02:
@tkenbo, thanks a lot!

closed 2023-11-02 by igaw