linux-nvme/libnvme #684: stack smashing detected with 'nvme sanitize-log /dev/nvme0'

igaw, 2023-07-26:
There are several reports that the sanitize log command is able to smash the stack somehow. 

https://github.com/storaged-project/udisks/issues/1152

For tracking and awareness reasons, I'll open here an issue here too.

igaw, 2023-07-26:
FWIW, I am not able to reproduce it and I really wonder how it is actually possible to achieve this. It looks all sane when poking with gdb at the addresses and alignments.

So next I'd like to check if there is a way when the firmware misbehaves, it could result in overwriting the stack. I assume this should be really really hard to do but who knows. @keithbusch in case you have some input on this, it would highly appreciated.

igaw, 2023-09-20:
With linux-nvme/nvme-cli#2051 merged, this problem should be solved.

gyakovlev, 2023-09-24:
Hi, are you planning tagging release anytime soon?
this is a bit awkward to backport to 1.5 and it affects libblockdev and udisks a lot.

ReillyBrogan, 2023-09-24:
> Hi, are you planning tagging release anytime soon? this is a bit awkward to backport to 1.5 and it affects libblockdev and udisks a lot.

On Solus we encountered several issues after updating libblockdev/udisks and we have all of them resolved with [this libnvme patch](https://github.com/linux-nvme/libnvme/commit/c56910f807795528fff7ba6b81f8efcdb4babe98) and the latest release versions of libblockdev and udisks. We currently have no reported udisks crash issues, so we're not going to bother backporting the linked PR and will instead wait for it to be in a release.

bulletmark, 2023-09-24:
@ReillyBrogan I experience this repeatable bug on Arch Linux. I built a new libnvme package with that patch applied and find it does not fix this issue.

Note: Arch updates just prior to me seeing this bug were:
```
2023-09-21 07:23:34 libnvme            1.5-1 installed
2023-09-21 07:23:34 libblockdev        2.28-4 -> 3.0.3-4
2023-09-21 07:23:34 udisks2            2.9.4-4 -> 2.10.1-1
```
Since then:
```
$ journalctl -qg 'stack smashing detected'
Sep 21 07:27:30 lt udisksd[1244]: *** stack smashing detected ***: terminated
Sep 22 07:27:24 lt udisksd[1250]: *** stack smashing detected ***: terminated
Sep 22 07:28:50 lt udisksd[859]: *** stack smashing detected ***: terminated
Sep 22 07:29:16 lt udisksd[1233]: *** stack smashing detected ***: terminated
Sep 22 20:57:44 lt udisksd[1351]: *** stack smashing detected ***: terminated
Sep 22 22:57:28 lt udisksd[859]: *** stack smashing detected ***: terminated
Sep 22 22:57:54 lt udisksd[1236]: *** stack smashing detected ***: terminated
Sep 22 22:59:55 lt udisksd[1351]: *** stack smashing detected ***: terminated
Sep 22 23:00:38 lt udisksd[1066]: *** stack smashing detected ***: terminated
Sep 22 23:01:04 lt udisksd[1441]: *** stack smashing detected ***: terminated
Sep 22 23:01:43 lt udisksd[1794]: *** stack smashing detected ***: terminated
Sep 22 23:04:21 lt udisksd[1890]: *** stack smashing detected ***: terminated
Sep 23 07:24:53 lt udisksd[900]: *** stack smashing detected ***: terminated
Sep 23 07:25:19 lt udisksd[1278]: *** stack smashing detected ***: terminated
Sep 23 07:29:48 lt udisksd[920]: *** stack smashing detected ***: terminated
Sep 23 11:02:14 lt udisksd[1004]: *** stack smashing detected ***: terminated
Sep 23 12:22:34 lt udisksd[1030]: *** stack smashing detected ***: terminated
Sep 23 12:22:59 lt udisksd[1340]: *** stack smashing detected ***: terminated
Sep 23 13:58:14 lt udisksd[1005]: *** stack smashing detected ***: terminated
Sep 23 13:58:39 lt udisksd[1360]: *** stack smashing detected ***: terminated
Sep 24 19:55:17 lt udisksd[1012]: *** stack smashing detected ***: terminated
Sep 24 19:55:43 lt udisksd[1375]: *** stack smashing detected ***: terminated
```

ReillyBrogan, 2023-09-24:
> @ReillyBrogan I experience this repeatable bug on Arch Linux. I built a new libnvme package with that patch applied and find it does not fix this issue.

The other thing you can try is to build libnvme from git trunk and see if that fixes the issue. If it does then you can bisect your way to the patch or series of patches that fixes it. If not then it's worth opening a new issue for it (please include the stack trace which was probably captured by systemd-coredump). Most likely you should open that issue in libblockdev.

gyakovlev, 2023-09-24:
> On Solus we encountered several issues after updating libblockdev/udisks and we have all of them resolved with [this libnvme [...]

thanks for the tip, that single patch helped. that should be good enough to stop endless udisk crash loop (which blocks entire desktop session) until next release drops.

igaw, 2023-09-25:
> Hi, are you planning tagging release anytime soon?

Yes, the next release is planned for end of this week. 

https://github.com/linux-nvme/nvme-cli/milestones

tbzatek, 2023-10-10:
Another case with either Seagate Firecuda 520 (Phison E16, 1bb1:5016) or Firecuda 530 (Phison E18, 1bb1:5018). Fixed by firmware update: https://github.com/storaged-project/udisks/issues/1195#issuecomment-1754105027

igaw, 2023-10-10:
I discussed this with hch offline and he said we should just send quirk patches which set the minimal DMA alignment requirement for these devices. I just didn't find time to write these patches yet.

keithbusch, 2023-10-11:
Wait a sec, wasn't the experiments revealing that the alignment didn't matter at all and the devices were simply overrunning the log page no matter what alignment we used? You're just going to corrupt kernel memory instead of user memory if you simply bounce.

igaw, 2023-10-11:
Yes, but if I understood Christoph right, he's point was that we send from kernel context commands (id ctrl, id ns) with the potentially `wrong` alignments for these devices. This part should addressed.

keithbusch, 2023-10-11:
Perhaps I'm missing something. Did any experiment show an alignment issue with any device?

igaw, 2023-10-11:
No, if I remember correctly. But the amount of data transferred was alignment dependent. We just got more devices reported which show this problem with ns ctrl and ns id commands issued from the traverse topology code. The kernel issues these commands as well, so it's not completely unlikely that we have a problem there as well. Or do I miss something?

igaw, 2023-10-31:
Too bad. Anyway, I am pondering how to add some logging code to the passthru path which checks the alignment and issues a warning. I am tracking this in https://github.com/linux-nvme/libnvme/issues/728. So let's close this one for now.

closed 2023-10-31 by igaw