linux-nvme/libnvme #700: Unable to build "nvme-cli" from "master" with "master"

SchweinDeBurg, 2023-08-19:
After commit f07b63103878c8c973182bf63459e2e1556cc2db "ioctl: remove nsid from nvme_identify_secondary_ctrl_list()" nvme-cli utility failed to build:
```
FAILED: nvme@exe/nvme-wrap.c.o 
ccache /usr/bin/gcc -Invme@exe -I. -I.. -Iccan -I../ccan -I/usr/include/json-c5 -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -std=gnu99 -fomit-frame-pointer -D_GNU_SOURCE -include config.h -O2 -g -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -m64 -march=native -mtune=native -fasynchronous-unwind-tables -mno-aes -w -MD -MQ 'nvme@exe/nvme-wrap.c.o' -MF 'nvme@exe/nvme-wrap.c.o.d' -o 'nvme@exe/nvme-wrap.c.o' -c ../nvme-wrap.c
../nvme-wrap.c: In function 'nvme_cli_identify_secondary_ctrl_list':
../nvme-wrap.c:24:10: error: too many arguments to function 'nvme_identify_secondary_ctrl_list'
   24 |   __rc = nvme_ ## op(d->direct.fd, __VA_ARGS__);  \
      |          ^~~~~
../nvme-wrap.c:113:9: note: in expansion of macro 'do_admin_op'
  113 |  return do_admin_op(identify_secondary_ctrl_list, dev, nsid, ctrl_id,
      |         ^~~~~~~~~~~
In file included from /usr/include/nvme/linux.h:14,
                 from /usr/include/libnvme.h:18,
                 from ../nvme-wrap.c:9:
/usr/include/nvme/ioctl.h:765:19: note: declared here
  765 | static inline int nvme_identify_secondary_ctrl_list(int fd,
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```
I've created the following temporary patch for myself:
```
commit f07b63103878c8c973182bf63459e2e1556cc2db
Author: Caleb Sander <csander@purestorage.com>
Date:   Sat Aug 12 13:51:18 2023 -0600

    ioctl: remove nsid from nvme_identify_secondary_ctrl_list()
    
    According to the NVMe specification, Identify CNS value 15h
    ("Secondary Controller list of controllers associated with
    the primary controller processing the command")
    does not use the NSID field.
    So remove the "nsid" argument from nvme_identify_secondary_ctrl_list().
    
    Signed-off-by: Caleb Sander <csander@purestorage.com>

diff --git b/src/nvme/ioctl.h a/src/nvme/ioctl.h
index 8ce80a0..4d843bc 100644
--- b/src/nvme/ioctl.h
+++ a/src/nvme/ioctl.h
@@ -748,6 +748,7 @@ static inline int nvme_identify_primary_ctrl(int fd, __u16 cntid,
 /**
  * nvme_identify_secondary_ctrl_list() - Retrieves secondary controller list
  * @fd:		File descriptor of nvme device
+ * @nsid:	Namespace identifier
  * @cntid:	Return controllers starting at this identifier
  * @sc_list:	User space destination address to transfer the data
  *
@@ -762,7 +763,7 @@ static inline int nvme_identify_primary_ctrl(int fd, __u16 cntid,
  * Return: The nvme command status if a response was received (see
  * &enum nvme_status_field) or -1 with errno set otherwise.
  */
-static inline int nvme_identify_secondary_ctrl_list(int fd,
+static inline int nvme_identify_secondary_ctrl_list(int fd, __u32 nsid,
 			__u16 cntid, struct nvme_secondary_ctrl_list *sc_list)
 {
 	struct nvme_identify_args args = {
@@ -773,7 +774,7 @@ static inline int nvme_identify_secondary_ctrl_list(int fd,
 		.timeout = NVME_DEFAULT_IOCTL_TIMEOUT,
 		.cns = NVME_IDENTIFY_CNS_SECONDARY_CTRL_LIST,
 		.csi = NVME_CSI_NVM,
-		.nsid = NVME_NSID_NONE,
+		.nsid = nsid,
 		.cntid = cntid,
 		.cns_specific_id = NVME_CNSSPECID_NONE,
 		.uuidx = NVME_UUID_NONE,
diff --git b/test/test.c a/test/test.c
index e41fc9d..2f24e1e 100644
--- b/test/test.c
+++ a/test/test.c
@@ -112,7 +112,7 @@ static int test_ctrl(nvme_ctrl_t c)
 		printf("  PASSED: Identify Primary\n");
 	else
 		printf("  ERROR: Identify Primary:%x\n", ret);
-	ret = nvme_identify_secondary_ctrl_list(fd, 0, &sec);
+	ret = nvme_identify_secondary_ctrl_list(fd, 1, 0, &sec);
 	if (!ret)
 		printf("  PASSED: Identify Secondary\n");
 	else
diff -up libnvme-1.5-git/test/ioctl/identify.c.orig libnvme-1.5-git/test/ioctl/identify.c
--- libnvme-1.5-git/test/ioctl/identify.c.orig	2023-08-17 15:58:39.000000000 +0300
+++ libnvme-1.5-git/test/ioctl/identify.c	2023-08-19 10:29:30.281771702 +0300
@@ -363,7 +363,7 @@ static void test_secondary_ctrl_list(voi
 
 	arbitrary(&expected_id, sizeof(expected_id));
 	set_mock_admin_cmds(&mock_admin_cmd, 1);
-	err = nvme_identify_secondary_ctrl_list(TEST_FD, TEST_CNTID, &id);
+	err = nvme_identify_secondary_ctrl_list(TEST_FD, NVME_NSID_NONE, TEST_CNTID, &id);
 	end_mock_cmds();
 	check(err == 0, "identify returned error %d, errno %m", err);
 	cmp(&id, &expected_id, sizeof(id), "incorrect identify data");
```

igaw, 2023-08-19:
Thanks for the report. Completely forgot to ask @calebsander to update nvme-cli as well.

closed 2023-08-19 by igaw

calebsander, 2023-08-19:
Sorry about that! It occurred to me that the change might break `nvme-cli` but I searched for the function name and didn't find any callers. Didn't consider that there might be a macro involved. I should have tried building it.