linux-nvme/libnvme #728: Use __nvme_alloc helper to allocate payload

igaw, 2023-10-10:
To ensure the payloads buffer allocation are correctly aligned use the newly added `__nvme_alloc` helper. 

See also https://github.com/linux-nvme/nvme-cli/pull/2051

tbzatek, 2023-10-12:
Some more fixes: #731 

Not sure about `nvme_get_telemetry_log()` as the buffer gets realloc'd and there's more stuff in `mi.c` that I was not sure about either (`nvme_mi_ep_probe()`).

igaw, 2023-10-12:
Thanks. I am also not sure about `mi.c`. We could also introduce the realloc helper from nvme-cli if needed.

Another thing we could do is add a check to `nvme_passwthru64` which issue a warning if the alignment is not on a page.

igaw, 2024-07-03:
The kernel learned to handle unaligned data structures.  So there is no need for this anymore.

closed 2024-07-03 by igaw