linux-nvme/libnvme #805: Question modified about nvme_id_ns_flbas_to_lbaf_inuse() in src/nvme/util.h

benchuanggli, 2024-04-01:
I have a question about modifying [8db4b1f37e17a3b6f060311624655246563b553b ](https://github.com/linux-nvme/libnvme/commit/8db4b1f37e17a3b6f060311624655246563b553b#diff-a145a5bb62273071b1691734777b1b3a9a08bb2c9ec8b2a99c24b04081ea6450)  the function nvme_id_ns_flbas_to_lbaf_inuse().

```
diff --git a/src/nvme/util.h b/src/nvme/util.h
index 0739cf2f..efdf9757 100644
--- a/src/nvme/util.h
+++ b/src/nvme/util.h
@@ -449,8 +449,8 @@ static inline void nvme_feature_decode_namespace_write_protect(__u32 value,
 
 static inline void nvme_id_ns_flbas_to_lbaf_inuse(__u8 flbas, __u8 *lbaf_inuse)
 {
-       *lbaf_inuse = (((flbas & NVME_NS_FLBAS_HIGHER_MASK) >> 1) |
-                       (flbas & NVME_NS_FLBAS_LOWER_MASK));
+       *lbaf_inuse = ((NVME_FLBAS_HIGHER(flbas) >> 1) |
+                       NVME_FLBAS_LOWER(flbas));
 }
```

After macro expansion about the FLBAS_HIGHER part
```
#define NVME_GET(value, name) \
        (((value) >> NVME_##name##_SHIFT) & NVME_##name##_MASK)

NVME_FLBAS_HIGHER(flbas) => 
=> NVME_GET(flbas, FLBAS_HIGHER)
=>  (((flbas) >> NVME_FLBAS_HIGHER_SHIFT) & NVME_FLBAS_HIGHER_MASK)

```

They should be unequal, right?
Is the previous one correct?

Thanks for your read.

igaw, 2024-04-02:
Yes, you are right. Sorry, I didn't catch this one.

closed 2024-04-02 by igaw

benchuanggli, 2024-04-02:
Maybe we need a test case for it https://github.com/linux-nvme/libnvme/pull/808.