linux-nvme/libnvme #832: Remove generated artificats from release?

igaw, 2024-05-03:
Currently, libnvme and also nvme-cli include generated artifacts such as the man pages into the release. I suspect the early days of the project this made sense, but these days I expect that most users get the library and cli via a package manager. I expect that most distros are building the documentation on their own. Thus I don't think we need to include these artifacts anymore. 

This would make the release a bit cleaner as we don't update the version string in `meson.build` and then have to build the artifacts and do another commit on top.

Thoughts?

tbzatek, 2024-05-03:
Agree. We do rebuild docs and manpages in Fedora and RHEL via `-Ddocs=all -Ddocs-build=true`.

However I see Gentoo ebuilds are just installing bundled docs, only supplying `-Ddocs=all` to meson.

Perhaps `-Ddocs` should always imply `-Ddocs-build` ?

igaw, 2024-05-03:
Yes, that is one of the consequences when we don't ship any prebuild documentation. 

Maybe we should first reach out to the major distros and telling about the plan.

igaw, 2024-05-06:
Got it and I understand it's hard sell. But so far we don't release a dist archive or is there some github magic going on?

Thanks for the links. I think we could do something like this based on the current meson instruction fairly simple. Though I wonder if this a good strategy in the long run with the recent xz exploit, meaning we have in the dist file data which is not part of the git tree. Or do I see ghosts?

thesamesam, 2024-05-12:
Right - so at the moment, we use the data from the git repo as it's committed. I understood the proposal here to be to stop doing that.

It's a fair question about dist vs git being different. On the upside, the diff should be easy to validate and it's not goops of scripts we execute...

I have a preference but I can totally understand if you don't feel comfortable with it too.

igaw, 2024-05-13:
I don't think we have to rip it out just  because I have a tine itch with the release process. 

Currently, the steps is to do the release by updating the version in the meson.build and then run the doc generate step. The reason for this order is that the docs contain the version string derived from the meson.build file. 

Maybe we should just define a version string for the docs separately and then reorder the release steps. Generate the docs first and the do the tag etc.

closed 2024-06-14 by igaw

pabs3, 2025-07-24:
Distros strongly prefer to have generated files excluded from the VCS/tarballs. For eg the [Debian advice](https://wiki.debian.org/AutoGeneratedFiles) says this, and documents several compromises.

thesamesam, 2025-07-24:
No, _Debian_ prefers that. See my remarks at https://github.com/iputils/iputils/pull/520#issuecomment-3111706575 as well.

igaw, 2025-07-24:
> Distros strongly prefer to have generated files excluded from the VCS/tarballs. For eg the [Debian advice](https://wiki.debian.org/AutoGeneratedFiles) says this, and documents several compromises.

Did you [read](https://github.com/linux-nvme/libnvme/issues/832#issuecomment-2106118747)? It's just not that simple. While I agree we should aim for source only, sometimes it's not possible.

PR #847 (2024-06-14): Change commit order on release
Let's reorder the release commits, so that the last commit is actually the version string update. The only reason for the previous order was the version string was used in the documentation. Though there are none, only the date is used.

Fixes: #832