linux-nvme/libnvme #840: Double free on scandir failure

smithkyl, 2024-05-16:
Running Debian 12 (nvme-cli = 2.4+really2.3-3, libnvme = 1.3-1), but reading through the code here, I believe this still applies.

nvme discover command failed with "free(): double free detected in tcache 2"

Backtrace (and some quick lookup):
```
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at ./nptl/pthread_kill.c:44
#1  0x00007f2a6f42be8f in __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  0x00007f2a6f3dcfb2 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007f2a6f3c7472 in __GI_abort () at ./stdlib/abort.c:79
#4  0x00007f2a6f420430 in __libc_message (action=action@entry=do_abort, fmt=fmt@entry=0x7f2a6f53a459 "%s\n") at ../sysdeps/posix/libc_fatal.c:155
#5  0x00007f2a6f4357aa in malloc_printerr (str=str@entry=0x7f2a6f53d098 "free(): double free detected in tcache 2") at ./malloc/malloc.c:5660
#6  0x00007f2a6f437a36 in _int_free (av=0x7f2a6f573c60 <main_arena>, p=0x55bf4b4e1860, have_lock=have_lock@entry=0) at ./malloc/malloc.c:4469
#7  0x00007f2a6f439e8f in __GI___libc_free (mem=mem@entry=0x55bf4b4e1870) at ./malloc/malloc.c:3385
#8  0x00007f2a6f5d266e in nvme_free_dirents (i=<optimized out>, d=0x55bf4b4e1870) at ../src/nvme/tree.c:55
#9  nvme_ctrl_scan_namespaces (r=r@entry=0x55bf4b4e0750, c=c@entry=0x55bf4b4e1950) at ../src/nvme/tree.c:1148
#10 0x00007f2a6f5d2bc4 in nvme_scan_ctrl (r=r@entry=0x55bf4b4e0750, name=0x55bf4b4e89c3 "nvme10") at ../src/nvme/tree.c:1461
#11 0x00007f2a6f5d2ea6 in nvme_scan_topology (r=r@entry=0x55bf4b4e0750, f=f@entry=0x0, f_args=f_args@entry=0x0) at ../src/nvme/tree.c:95
#12 0x000055bf49f16401 in nvmf_discover (desc=0x55bf49f9e330 "Send Get Log Page request to Discovery Controller.", argc=5, argv=<optimized out>, connect=<optimized out>) at ../fabrics.c:830
#13 0x000055bf49f51b35 in handle_plugin (argc=argc@entry=5, argv=argv@entry=0x7ffd493c3190, plugin=0x55bf49fe4a40 <builtin>) at ../plugin.c:164
#14 0x000055bf49f143e6 in main (argc=6, argv=0x7ffd493c3188) at ../nvme.c:8963

(gdb) frame 9
#9  nvme_ctrl_scan_namespaces (r=r@entry=0x55bf4b4e0750, c=c@entry=0x55bf4b4e1950) at ../src/nvme/tree.c:1148
(gdb) print ret
$1 = -1
```
From this data, we can tell that the scandir call (nvme_scan_ctrl_namespaces) failed. scandir doesn't guarantee a NULL pointer on failure even though it may have modified it. Thus, I believe nvme_free_dirents attempted to (re)free an invalid list pointer, causing a process abort. While the code I'm running isn't using the cleanup macros, I haven't seen anything in their documentation that would lead me to believe that they behave differently.

As a fix, I think that scandir callers or scandir-wrapper callers just need to force the list pointer to NULL at some point after failure and before clean up or just avoid cleaning up after failure in older versions.

smithkyl, 2024-05-23:
Was wondering if something like the following might be a reasonable fix for this issue:
```
--- a/src/nvme/filters.c
+++ b/src/nvme/filters.c
@@ -74,34 +74,45 @@ int nvme_subsys_filter(const struct dirent *d)
        return 0;
 }

+static int __nvme_scandir(const char *dirp, struct dirent ***namelist,
+                         int (*filter)(const struct dirent *),
+                         int (*compar)(const struct dirent **,
+                                       const struct dirent **))
+{
+       int ret = scandir(dirp, namelist, filter, compar);
+       if( ret < 0 )
+               *namelist = NULL;
+       return ret;
+}
+
 int nvme_scan_subsystems(struct dirent ***subsys)
 {
        const char *dir = nvme_subsys_sysfs_dir();

-       return scandir(dir, subsys, nvme_subsys_filter, alphasort);
+       return __nvme_scandir(dir, subsys, nvme_subsys_filter, alphasort);
 }

 int nvme_scan_subsystem_namespaces(nvme_subsystem_t s, struct dirent ***ns)
 {
-       return scandir(nvme_subsystem_get_sysfs_dir(s), ns,
-                      nvme_namespace_filter, alphasort);
+       return __nvme_scandir(nvme_subsystem_get_sysfs_dir(s), ns,
+                             nvme_namespace_filter, alphasort);
 }

 int nvme_scan_ctrls(struct dirent ***ctrls)
 {
        const char *dir = nvme_ctrl_sysfs_dir();

-       return scandir(dir, ctrls, nvme_ctrls_filter, alphasort);
+       return __nvme_scandir(dir, ctrls, nvme_ctrls_filter, alphasort);
 }

 int nvme_scan_ctrl_namespace_paths(nvme_ctrl_t c, struct dirent ***paths)
 {
-       return scandir(nvme_ctrl_get_sysfs_dir(c), paths,
-                      nvme_paths_filter, alphasort);
+       return __nvme_scandir(nvme_ctrl_get_sysfs_dir(c), paths,
+                             nvme_paths_filter, alphasort);
 }

 int nvme_scan_ctrl_namespaces(nvme_ctrl_t c, struct dirent ***ns)
 {
-       return scandir(nvme_ctrl_get_sysfs_dir(c), ns,
-                      nvme_namespace_filter, alphasort);
+       return __nvme_scandir(nvme_ctrl_get_sysfs_dir(c), ns,
+                             nvme_namespace_filter, alphasort);
```
I think this should fix the issue above as well as provide a test point where faking a scandir() failure could reproduce the issue more easily.

igaw, 2024-05-23:
Could you test if the current upstream version has this problem? I think we address it already with ba533deb2155 ("tree: use cleanup functions")

smithkyl, 2024-05-23:
I had looked at ba533de and while the struct initialization does prevent an uninitialized pointer from being freed on early failures, I was still concerned about mid-scandir()-execution failures (ENOMEM-type situations). None of the documentation around scandir() seems to indicate what *namelist's value might be on failure which could indicate that the value is indeterminate/unspecified since anything it might point to should be freed anyway. With that potential for failure in mind, I ended up pulling glibc's source and found that scandir() only modifies *namelist on success instead of (re)allocating memory directly into it during processing. As such, at least for the time being, this should be fixed.

That all said, it seems like there is still some dependency on undocumented behavior and implementation details here. But, given that glibc probably wouldn't change a detail like that, I'm fine with closing this out and filing something with the Debian folks if folks here are fine with that dependency (or if they have documentation I haven't found that guarantees current behavior).

igaw, 2024-05-24:
Generally, a function should only update the output arguments on success, IMO. Unfortunately, POSIX is a mix bag,  some function may update the output even on failure, e.g. strtol according my quick search.

Anyway,  I don't think glibc will change the behavior for scandir. And we also depend on this in various other places. I was also unable to find any documentation on this topic for glibc.

Thanks for double checking the glibc source. I agree I don't expect any behavior change in glibc, actually I wouldn't surprised if glibc is actually guaranteeing this. But this would be a question for an expert.

closed 2024-05-24 by igaw