linux-nvme/libnvme #875: Unable to insert TLS v1 key into the kernel keyring

martin-gpy, 2024-08-05:
After the recent libnvme commit to update PSK derivation for TLS v1 (as part of https://github.com/linux-nvme/libnvme/commit/854a6dcb38738a60a016c83b4736bfd74f07b208), one is unable to insert the TLS v1 key into the kernel keyring as shown below:

SLES15 SP6 FCS:
```# uname -r```
```6.4.0-150600.23.7-default```

```# nvme --version```
```nvme version 2.10 (git 2.10-2-g9dd886a)```
```libnvme version 1.10 (git 1.10-1-g67d7d75)```

Now proceeding to insert a TLS v1 key into the kernel keyring, one can see that fails (i.e. TLS key 00000000):

```# nvme gen-tls-key -I 1 -m 1 -c nqn.1992-08.com.netapp:sn.0a86ed924b8d11efa459d039ea482fba:subsystem.s2 -i```
```NVMeTLSkey-1:01:uE2MWlr4JUEhXSYi++9FsPuWa69Ud9w0Dhx6YkS7HhR88s+F:```
```Inserted TLS key 00000000```

But at the same time, one can continue to insert a TLS v0 key successfully without any issues:

```# nvme gen-tls-key -I 0 -m 1 -c nqn.1992-08.com.netapp:sn.0a86ed924b8d11efa459d039ea482fba:subsystem.s2 -i```
```NVMeTLSkey-1:01:CiDHhmC6uZ6WYOWoRYNeQpxbcUjOmRe5X24jtmlbtvqeVe9t:```
```Inserted TLS key 35c10fe8```

igaw, 2024-08-05:
The `key_len` is hard coded in `gen_tls_key`, whereas the in `gen_dhchap_key` has some code to figure out the key length. Clearly, `gen_tls_key` needs to be extended to get this new feature working.

martin-gpy, 2024-08-07:
Should be fixed with the pull request at https://github.com/linux-nvme/libnvme/pull/877

igaw, 2024-08-07:
Merged the fix. Though I think we  should open an new issue for the missing support for the different `hmac`. `gen-tls-key` only supports `-m 1`.

closed 2024-08-07 by igaw

prashanth-nayak, 2024-08-07:
> Merged the fix. Though I think we should open an new issue for the missing support for the different `hmac`. `gen-tls-key` only supports `-m 1`.

I did test with '-m 2' and it worked:

ssan-rx2530-15:~ # nvme gen-tls-key -n nqn.2014-08.org.nvmexpress:uuid:713b3816-f9bf-ba43-b95a-5e4bf8c726e9 -c nqn.1992-08.com.netapp:sn.0a86ed924b8d11efa459d039ea482fba:subsystem.stest -m 2 -I 1 -i
NVMeTLSkey-1:02:nW/4y/TdCoDbgBE79IkRPNaeJapY7zeVHSIXFO1deBWd08f5wY1lhb2+HqJMnsYpMuBbSA==:
Inserted TLS key 0b4b0d17
ssan-rx2530-15:~ # cat /proc/keys | grep NVMe
0b4b0d17 I--Q---     1 perm 3b010000     0     0 psk       NVMe1R02 nqn.2014-08.org.nvmexpress:uuid:713b3816-f9bf-ba43-b95a-5e4bf8c726e9 nqn.1992-08.com.netapp:sn.0a86ed924b8d11efa459d039ea482fba:subsystem.stest APJi3AdhxbtXtPqPKLxgH7K02w27HL4TPo5k/hPpXMCEGs/yx4VH0rOOvAllnLw0: 48