linux-nvme/libnvme #909: 'make rpm' puts files in /usr/local by default.

BigValen, 2024-11-07:
```
#   make rpm
git archive --format=tar HEAD > libnvme.tar
tar rf libnvme.tar .build/libnvme.spec
gzip -f -9 libnvme.tar
rpmbuild -ta libnvme.tar.gz -v
setting SOURCE_DATE_EPOCH=1657670400
Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.1ObXeU
+ umask 022
+ cd /root/rpmbuild/BUILD
+ cd /root/rpmbuild/BUILD
+ rm -rf libnvme-1.11
+ /usr/bin/mkdir -p libnvme-1.11
+ cd libnvme-1.11
+ /usr/bin/gzip -dc /libnvme/libnvme.tar.gz
+ /usr/bin/tar -xof -
+ STATUS=0
+ '[' 0 -ne 0 ']'
+ /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
+ RPM_EC=0
++ jobs -p
+ exit 0
Executing(%build): /bin/sh -e /var/tmp/rpm-tmp.F5hhP0
+ umask 022
+ cd /root/rpmbuild/BUILD
+ cd libnvme-1.11
+ meson .build -Ddocs=man -Ddocs-build=true -Ddefault_library=both
The Meson build system
Version: 0.63.3
Source dir: /root/rpmbuild/BUILD/libnvme-1.11
Build dir: /root/rpmbuild/BUILD/libnvme-1.11/.build
Build type: native build
Project name: libnvme
Project version: 1.11
C compiler for the host machine: cc (gcc 11.4.1 "cc (GCC) 11.4.1 20231218 (Red Hat 11.4.1-3)")
C linker for the host machine: cc ld.bfd 2.35.2-24
Host machine cpu family: x86_64
Host machine cpu: x86_64
C++ compiler for the host machine: c++ (gcc 11.4.1 "c++ (GCC) 11.4.1 20231218 (Red Hat 11.4.1-3)")
C++ linker for the host machine: c++ ld.bfd 2.35.2-24
Found pkg-config: /usr/bin/pkg-config (1.7.3)
Run-time dependency json-c found: YES 0.14
Run-time dependency openssl found: YES 3.0.7
Header "openssl/opensslv.h" has symbol "LIBRESSL_VERSION_NUMBER" with dependency openssl: NO
Has header "openssl/core_names.h" with dependency openssl: YES
Run-time dependency libkeyutils found: YES 1.6.3
Checking if "__builtin_type_compatible_p" compiles: YES
Checking if "typeof" compiles: YES
Checking if "byteswap.h" compiles: YES
Checking if "bswap64" : links: YES
Checking if "statement-expr" compiles: YES
Checking if "isblank" : links: YES
Checking if "linux/mctp.h" compiles: NO
Checking if "netdb" : links: YES
Checking if "ioctl has glibc-style prototype" compiles: YES
Compiler for C supports function attribute fallthrough: YES
Configuring libnvme.spec using configuration
Configuring config.h using configuration
Program python3 found: YES (/usr/bin/python3)
Program swig found: YES (/usr/bin/swig)
Has header "Python.h" with dependency python: NO
Configuring nbft-dump-diff.sh using configuration
Configuring gen-nbft-diffs.sh using configuration
Program diff found: YES (/usr/bin/diff)
Program diff found: YES (/usr/bin/diff)
Program tree-diff.sh found: YES (/bin/bash -e /root/rpmbuild/BUILD/libnvme-1.11/test/sysfs/tree-diff.sh)
Program diff found: YES (/usr/bin/diff)
Program config-diff.sh found: YES (/root/rpmbuild/BUILD/libnvme-1.11/test/config/config-diff.sh)
Configuring conf.py using configuration
Configuring api.rst using configuration
Configuring index.rst using configuration
Configuring quickstart.rst using configuration
Configuring installation.rst using configuration
Configuring mi.rst using configuration
Configuring config-schema.json using configuration
Program /root/rpmbuild/BUILD/libnvme-1.11/doc/rst/../../scripts/kernel-doc found: YES (/root/rpmbuild/BUILD/libnvme-1.11/doc/rst/../../scripts/kernel-doc)
Program /root/rpmbuild/BUILD/libnvme-1.11/doc/../scripts/kernel-doc found: YES (/root/rpmbuild/BUILD/libnvme-1.11/doc/../scripts/kernel-doc)
Program /root/rpmbuild/BUILD/libnvme-1.11/doc/../scripts/kernel-doc-check found: YES (/root/rpmbuild/BUILD/libnvme-1.11/doc/../scripts/kernel-doc-check)
Program /root/rpmbuild/BUILD/libnvme-1.11/doc/../scripts/list-man-pages.sh found: YES (/root/rpmbuild/BUILD/libnvme-1.11/doc/../scripts/list-man-pages.sh)
Build targets in project: 939

libnvme 1.11

  Paths
    prefixdir      : /usr/local
    sysconfdir     : /usr/local/etc
    bindir         : /usr/local/bin
    includedir     : /usr/local/include
    datadir        : /usr/local/share
    mandir         : share/man/man2
    libdir         : /usr/local/lib64
    build location : /root/rpmbuild/BUILD/libnvme-1.11/.build

  Dependencies
    json-c         : True
    OpenSSL        : True
    keyutitls      : True
    libdbus        : False
    Python 3       : True
```

Five minutes later ...

```
+ exit 0

RPM build errors:
    File not found: /root/rpmbuild/BUILDROOT/libnvme-1.11-0.x86_64/usr/lib64/libnvme*
```

This can be fixed with:

```
sed -i '/meson \.build/s/$/ --prefix=\/usr  -Dprefer_static=true/' libnvme.spec.in
```

Before I make up a simple PR...is that intentional ? Or did meson's behaviour change recently ?

igaw, 2024-11-08:
The Makefile is not really supported. We switch to meson and the Makefile is just a wrapper around some common targets. It exists is for those of use, who have 'make' in their fingers automated. 

The rpm target was originally added to be able to ship easily development version to testers. These days nvme is packaged the distros. IIRC, someone brought back the rpm target to some reason. We have an AppImage for this purpose nowadays as getting rpm working is not fun. 

The prefix is intentional not set to /usr (same as with autotools), to prevent overwriting the version shipped by the distro.

Anyway, packaging is not a trivial tasks and I came to the conclusion it's not something an upstream project should even attempt. There is no one size fits all. Provide a sane build tool chain (that removes autohell from the picture) and give the necessary knobs to get everything working. 

Long story short, I am not against changing the rpm build step, just note I am not really interested in maintaining it. So if someone screams after a change, I'll let you fix it. Deal?

sc108-lee, 2024-11-26:
it will be fixed with above PR

igaw, 2024-12-02:
I've merged the PR. Thanks a lot!

closed 2024-12-02 by igaw