linux-nvme/libnvme #930: release notes: mention mctp changes

igaw, 2024-12-13:
As far as I am aware, there is minimal external usage of the dbus-based device enumeration at present, so a changeover should have minimal impact. Might be worth a mention in the libnvme changelog though.

_Originally posted by @jk-ozlabs in https://github.com/linux-nvme/libnvme/issues/929#issuecomment-2540409008_

jk-ozlabs, 2024-12-16:
If you need some text for that:

> This release switches to the newer dbus API for `mctpd` (corresponding to `mctpd` version 2.0 and later). If you're using libnvme's MCTP-device endpoint enumeration functions (`nvme_mi_for_each_endpoint`), you will need to update to mctpd v2.x or later.

of course, feel free to adjust for suitability for the notes format.

igaw, 2025-04-03:
Finally updated the release notes...

closed 2025-04-03 by igaw