linux-nvme/libnvme #976: Python upload fails

igaw, 2025-03-26:
https://github.com/linux-nvme/libnvme/actions/runs/14068389590/job/39396515053

```
ERROR    InvalidDistribution: Invalid distribution metadata: unrecognized or  malformed field 'description'    
```

It seems that the twine tool chokes on one something.

igaw, 2025-03-26:
@glimchb any idea?

glimchb, 2025-03-26:
according to similar https://github.com/pypa/twine/issues/1216
this could help:
```
pip install -U packaging
```
before calling `pypa/gh-action-pypi-publish`

igaw, 2025-04-01:
I've dissabled the action for now as I was too stupid to get it working again.

igaw, 2025-04-10:
I've tried to use the virtual environment but the action does filter out any special paths. I'm giving up, can't waste more time into this. If someone wants to fix it, please be my guest. Otherwise, PyPi uploads are history.

glimchb, 2025-04-10:
@igaw let me try and fix it

igaw, 2025-04-11:
Thanks a lot! I got a bit frustrated with the Github Actions/Workflows. They are great when they work but if not, they are so time consuming to fix. It's difficult to justify the time it takes to get them working...

glimchb, 2025-04-11:
Agree! Sorry for not jumping on this sooner

glimchb, 2025-04-11:
@igaw publishing fails on `Error: Trusted publishing exchange failure`
can you fix the trusted publisher in https://test.pypi.org/manage/project/libnvme/settings/publishing/

[image]

igaw, 2025-04-11:
Thanks,  the final hurdle was the 'environment'. I alreaddy set it explicitly to 'pypi'. So after adding it, the test upload works again. I've replicated the settings for the release upload. But I don't think I can upload the v1.13 via the action now.

glimchb, 2025-04-11:
we can add and or statement like this `|| ${{ github.event_name == 'workflow_dispatch' }}` to this line https://github.com/linux-nvme/libnvme/blob/4c189213157c3d95687494b3c2cf013bf12e55f7/.github/workflows/release-python.yml#L63 if you want to...

igaw, 2025-04-14:
Wouldn't this mean we upload every single merge into master?

What we would be nice is a workflow trigger with an argument, e.g. build a tag with but using the latest workflow version.

igaw, 2025-04-14:
chatgpt says we would need something like

```yaml
name: Run workflow on a historical tag

on:
  workflow_dispatch:
    inputs:
      tag:
        description: 'Tag to check out'
        required: true

jobs:
  run-on-tag:
    runs-on: ubuntu-latest
    steps:
      - name: Fetch full history
        uses: actions/checkout@v4
        with:
          fetch-depth: 0

      - name: Checkout specific tag
        run: |
          git fetch --tags
          git checkout ${{ github.event.inputs.tag }}

      - name: Show current commit
        run: git log -1

      - name: Your steps here
        run: |
          echo "Running on tag: ${{ github.event.inputs.tag }}"
          # put your build/test/etc. here
```

From a quick look, it seems to make sense...

glimchb, 2025-04-14:
`workflow_dispatch` is exactly that: manual triggering

igaw, 2025-04-15:
Ah, I see. I didn't really realize what `workflow_dispatch` does. So that means the Python workflow needs to be triggered when adding a new tag and also when manually triggered to build. But in the second case the correct version needs to be checkout by hand. 

And there is another problem which the test upload runs into: it can't upload the same library version, so the test uplaod at least needs some new version number for every build. 

Let me reopen this issue then. Maybe I find some time to figure this out eventually.

igaw, 2025-04-15:
- [ ] allow development version upload to PyPi Tests 
- [ ] cleanup/remove development versions on PyPi Tests
       https://pypi.org/project/pypi-cleanup/
       https://github.com/arcivanov/pypi-cleanup
- [ ] `workflow_dispatch`: use input values from workflow_dispatch to build specified version

igaw, 2025-04-22:
I fixed the the PyPi Tests upload. The cleanup task still fails and the build from specific version thing is still missing.

This Python upload feature has used up so much time from me, that next time I'll just drop this feature if it starts to fail again. I will not maintain this.

glimchb, 2025-04-22:
@igaw Dell is happy to help out in this area, no need for you to spend time on this. Just reach for me @glimchb or @martin-belanger and we will jump on the issue when needed

igaw, 2025-04-24:
#1004 should allow to upload older releases but will fail for v1.12 v1.13 as these versions have the broken project toml file. I don't know how to fix this cleanly. I suppose v1.12 and v1.13 will never be uploaded to PyPi.

Next we need the cleanup tasks for the test PyPi so we don't run out of space.

PR #991 (2025-04-11): build: switch to new python meson backend
Fixes #976
commit 4c189213 (2025-04-11): build: switch to new python meson backend

Fixes #976