linux-nvme/libnvme #979: Bug Report: `nvme sanitize-log` returns zeroed output with libnvme 1.12 (works in 1.11.1)

mschabhuettl, 2025-04-02:
#### Summary

When using `nvme sanitize-log` with **libnvme 1.12**, the output shows all-zero values for progress, status, and estimated times. The same command works as expected with **libnvme 1.11.1**.

This appears to be a regression introduced in libnvme 1.12.

---

#### Environment

- `libnvme` version: 1.12 (git 1.12) → incorrect output  
- `libnvme` version: 1.11.1 (git 1.11.1) → correct output  
- `nvme-cli` versions tested: 2.12 and 2.11  
- OS: Arch Linux live ISO (`archlinux-2024.04.01-x86_64.iso`)  
- Device: `/dev/nvme1n1` (can provide full `nvme id-ctrl` if needed)

---

#### Steps to Reproduce

1. Boot into an Arch Linux live environment  
2. Use `nvme sanitize-log /dev/nvme1n1` on libnvme 1.12  
3. Observe that all values are `0` (progress, status, estimated time, etc.)  
4. Repeat with libnvme 1.11.1 – values are populated and accurate

---

#### Expected Behavior (libnvme 1.11.1)

```
Sanitize Progress (SPROG): 65535
Sanitize Status (SSTAT): 0x101
Sanitize Command Dword 10 Information (SCDW10): 0x2
Estimated Time For Block Erase: 4294967295 (No time provided)
...
```

#### Actual Behavior (libnvme 1.12)

```
Sanitize Progress (SPROG): 0
Sanitize Status (SSTAT): 0
Sanitize Command Dword 10 Information (SCDW10): 0
Estimated Time For Block Erase: 0
...
```

---

#### Screenshots

**libnvme 1.11.1 (correct):**  
[image]

**libnvme 1.12 (bug):**  
[image]

---

#### Notes

- This could be a parsing/decoding issue introduced in libnvme 1.12.  
- A sanitize operation had been started previously, so valid values were expected.

Let me know if I can provide a dump or logs to help with debugging. Thanks a lot for maintaining this project!

mschabhuettl, 2025-04-02:
Update:

I double-checked using `nvme get-log /dev/nvme1n1 --log-id=0x81 --log-len=512 -b` and received only zeroed data (all bytes are `0x00`), even though a sanitize operation had previously been started.

This reinforces that the issue likely lies in how libnvme 1.12 parses or requests the sanitize log page.

Interestingly, I noticed in the [v1.12 release notes](https://github.com/linux-nvme/libnvme/releases/tag/v1.12) that structured support for the sanitize log was newly added. So this might be related to the updated `nvme_sanitize_log_*()` implementation.

Let me know if I can help by providing raw dumps or more device info.

mschabhuettl, 2025-04-02:
Update: Confirmed regression is in libnvme 1.12

I just tested the following configuration:

- `nvme-cli 2.12` ✅
- `libnvme 1.11.1` ✅ → sanitize-log works correctly
- `libnvme 1.12` ❌ → sanitize-log returns all zero values

So the CLI version does *not* seem to be the issue — `nvme-cli 2.12` works perfectly when paired with `libnvme 1.11.1`.

This confirms that the regression or parsing issue is entirely within `libnvme 1.12`.

Let me know if you’d like help narrowing it down or testing a patch!

mschabhuettl, 2025-04-02:
Just to add a reference here:

This may be similar to [`nvme-cli` issue #2626](https://github.com/linux-nvme/nvme-cli/issues/2626), where newer versions of the CLI and `libnvme` explicitly check the `dalb3` field in the telemetry log header. If `dalb3 == 0`, the tool now returns an error (`No telemetry data block`), while older versions ignored that value and proceeded to read/log anyway.

I suspect the `sanitize-log` behavior in `libnvme 1.12` (or `nvme-cli 2.x`) is affected by a similar logic. The log header retrieved via `nvme get-log /dev/nvmeX --log-id=0x81` contains only zeros, so the function exits early or ignores the data entirely.

The "working" state in older `libnvme 1.11.1` may have just skipped that check and returned meaningless log content — just like in #2626.

If this is by design, it would be helpful to document or output a warning instead of silently returning all-zero logs.

Let me know if you'd like a sanitized hexdump or a direct comparison between both versions' behavior.

mschabhuettl, 2025-04-02:
Closing this issue.

The problem was due to using the namespace block device (`/dev/ng1n1`) instead of the controller device (`/dev/nvme1`) when running `sanitize-log`. The log appears empty when using the namespace, but works as expected on the controller.

Maybe worth adding a warning or clearer message when such logs are retrieved from a namespace instead of a controller.

Thanks for the help!

closed 2025-04-02 by mschabhuettl

igaw, 2025-04-02:
This is a bit strange, we had some changes how a device is opened but we reverted those and they were for nvme-cli. The current behavior of smart-log is correct and it is also documented in the man pages. Yes, I agree we could make these commands a bit more user friendly.

Your logs show though the different behavior was introduced by the library update. Could you check if your libnvme links against liburing?