linux-nvme/libnvme #995: Issue with Retrieving Namespace Information for Remote NVMe Controller

kkalbhor-git, 2025-04-21:
I hope this message finds you well.

I am currently working with the libnvme library and I have encountered an issue while retrieving the namespace details for an NVMe controller exposed over the network. My program correctly identifies the controllers but fails to retrieve the namespace details for a specific controller, nvme3, which is an NVMe-oF target.

Here is a minimal example of the code I am using:

```
#include <iostream>
#include <libnvme.h>
#include <stdio.h>
#include <stdlib.h>

void print_namespaces(nvme_ctrl_t ctrl)
{
    nvme_ns_t ns = nvme_ctrl_first_ns(ctrl);
    if (!ns) {
        std::cerr << "No namespaces found on controller " << nvme_ctrl_get_name(ctrl) << ".\n";
        return;
    }
    printf("Namespaces on controller %s:\n", nvme_ctrl_get_name(ctrl));
    while (ns) {
        struct nvme_id_ns id;
        int nsid = nvme_ns_get_nsid(ns);
        const char* ctrl_name = nvme_ctrl_get_name(ctrl);

        printf("Debug: Processing namespace NSID %d on controller %s\n", nsid, ctrl_name);

        if (nvme_ns_identify(ns, &id)) {
            std::cerr << "Failed to identify namespace NSID " << nsid << " on controller " << ctrl_name << ".\n";
            ns = nvme_ctrl_next_ns(ctrl, ns);
            continue;
        }

        unsigned lba_index = id.flbas & 0xf;
        uint32_t block_size = 1 << id.lbaf[lba_index].ds;
        uint64_t size_bytes = id.nsze * block_size;

        printf("    /dev/%sn%d\n", ctrl_name, nsid);
        printf("        NSID     : %d\n", nsid);
        printf("        Capacity : %.2f GB\n", size_bytes / (1024.0 * 1024 * 1024));
        printf("        Format   : %u B + %u B\n",
               block_size, id.lbaf[lba_index].ms);

        ns = nvme_ctrl_next_ns(ctrl, ns);
    }
}

void print_controller(nvme_ctrl_t ctrl)
{
    struct nvme_id_ctrl id;
    if (nvme_ctrl_identify(ctrl, &id)) {
        std::cerr << "Failed to identify controller.\n";
        return;
    }

    printf("Controller     : %s\n", nvme_ctrl_get_name(ctrl));
    printf("    SN         : %.20s\n", id.sn);
    printf("    MN         : %.40s\n", id.mn);
    printf("    FR         : %.8s\n", id.fr);

    print_namespaces(ctrl);
}

void print_subsystems(nvme_root_t root)
{
    nvme_host_t host = nvme_default_host(root);
    if (!host) {
        std::cerr << "Failed to get default host from NVMe tree.\n";
        return;
    }

    nvme_subsystem_t sub = nvme_first_subsystem(host);
    while (sub) {
        const char* nqn = nvme_subsystem_get_nqn(sub);
        printf("Subsystem NQN  : %s\n", nqn);

        nvme_ctrl_t ctrl = nvme_subsystem_first_ctrl(sub);
        while (ctrl) {
            print_controller(ctrl);
            ctrl = nvme_subsystem_next_ctrl(sub, ctrl);
        }

        sub = nvme_next_subsystem(host, sub);
        printf("\n");
    }
}

int main()
{
    nvme_root_t root = nvme_scan(NULL);
    if (!root) {
        std::cerr << "Failed to scan NVMe tree\n";
        return 1;
    }

    print_subsystems(root);
    nvme_free_tree(root);
    return 0;
}

```
Output:

```
Subsystem NQN  : nqn.2014-08.org.nvmexpress:uuid:52d06941-4112-18d7-77d4-36c91551f4f1
Controller     : nvme0
    SN         : VMware NVME_0002
    MN         : VMware Virtual NVMe Disk
    FR         : 1.3
Namespaces on controller nvme0:
Debug: Processing namespace NSID 1 on controller nvme0
    /dev/nvme0n1
        NSID     : 1
        Capacity : 16.00 GB
        Format   : 512 B + 0 B
... [similar output for other controllers] ...

Subsystem NQN  : KK
Controller     : nvme3
    SN         : 5d4207558844652906d7
    MN         : Linux
    FR         : 6.1.0-32
No namespaces found on controller nvme3.

```
However, when I run sudo nvme list -v, I can see that the nvme3 controller is visible and properly exposed over the network as follows:

```

Device            Cntlid SN                   MN                                       FR       TxPort Address        Slot   Subsystem    Namespaces
----------------- ----------------- ---------- -------------------------- ---------------- ---------------- ----------------
nvme0    0      VMware NVME_0002     VMware Virtual NVMe Disk                 1.3      pcie   0000:04:00.0   161    nvme-subsys0 nvme0n1, nvme0n2, nvme0n3
nvme1    0      VMware NVME_0000     VMware Virtual NVMe Disk                 1.3      pcie   0000:13:00.0   224    nvme-subsys1 nvme1n1, nvme1n2, nvme1n3
nvme2    0      VMware NVME_0001     VMware Virtual NVMe Disk                 1.3      pcie   0000:1b:00.0   256    nvme-subsys2 nvme2n1, nvme2n2, nvme2n3
nvme3    1      5d4207558844652906d7 Linux                                    6.1.0-32 tcp    traddr=172.16.206.236,trsvcid=4421,src_addr=172.16.206.236        nvme-subsys3 nvme3n1

```
As you can see, the namespaces for nvme3 are visible via the nvme list -v command, but my program is unable to retrieve the namespace information for the nvme3 controller. The other controllers' namespaces are successfully listed.

```
Version Information:
nvme-cli version: 2.13 (git 2.13-1-g4a66443)
libnvme version: 1.13 (git 1.13)
```

Could you please help me understand why this might be happening? Is there something I might be missing in terms of configuration or usage of libnvme for remote NVMe-oF controllers?

Thank you for your help!

Best regards,
Ketan Kalbhor
ketan.kalbhor@hitachivantara.com

kkalbhor-git, 2025-04-22:
Full nvme list -v output:

```

Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys0     nqn.2014-08.org.nvmexpress:uuid:52d06941-4112-18d7-77d4-36c91551f4f1                             nvme0
nvme-subsys1     nqn.2014-08.org.nvmexpress:uuid:52d010f9-d956-21bd-cf65-37ff35982640                             nvme1
nvme-subsys2     nqn.2014-08.org.nvmexpress:uuid:528efd65-5efb-1842-a6fd-a21505ed030b                             nvme2
nvme-subsys3     KK                                                                                               nvme3

Device           Cntlid SN                   MN                                       FR       TxPort Address        Slot   Subsystem    Namespaces
---------------- ------ -------------------- ---------------------------------------- -------- ------ -------------- ------ ------------ ----------------
nvme0    0      VMware NVME_0002     VMware Virtual NVMe Disk                 1.3      pcie   0000:04:00.0   161    nvme-subsys0 nvme0n1, nvme0n2, nvme0n3
nvme1    0      VMware NVME_0000     VMware Virtual NVMe Disk                 1.3      pcie   0000:13:00.0   224    nvme-subsys1 nvme1n1, nvme1n2, nvme1n3
nvme2    0      VMware NVME_0001     VMware Virtual NVMe Disk                 1.3      pcie   0000:1b:00.0   256    nvme-subsys2 nvme2n1, nvme2n2, nvme2n3
nvme3    1      5d4207558844652906d7 Linux                                    6.1.0-32 tcp    traddr=172.16.206.236,trsvcid=4421,src_addr=172.16.206.236        nvme-subsys3 nvme3n1

Device            Generic           NSID       Usage                      Format           Controllers
----------------- ----------------- ---------- -------------------------- ---------------- ----------------
/dev/nvme0n1 /dev/ng0n1   0x1         17.18  GB /  17.18  GB    512   B +  0 B   nvme0
/dev/nvme0n2 /dev/ng0n2   0x2         17.18  GB /  17.18  GB    512   B +  0 B   nvme0
/dev/nvme0n3 /dev/ng0n3   0x3         17.18  GB /  17.18  GB    512   B +  0 B   nvme0
/dev/nvme1n1 /dev/ng1n1   0x2         17.18  GB /  17.18  GB    512   B +  0 B   nvme1
/dev/nvme1n2 /dev/ng1n2   0x3         17.18  GB /  17.18  GB    512   B +  0 B   nvme1
/dev/nvme1n3 /dev/ng1n3   0x4         17.18  GB /  17.18  GB    512   B +  0 B   nvme1
/dev/nvme2n1 /dev/ng2n1   0x1         17.18  GB /  17.18  GB    512   B +  0 B   nvme2
/dev/nvme2n2 /dev/ng2n2   0x6         17.18  GB /  17.18  GB    512   B +  0 B   nvme2
/dev/nvme2n3 /dev/ng2n3   0x7         17.18  GB /  17.18  GB    512   B +  0 B   nvme2
/dev/nvme3n1 /dev/ng3n1   0x3         17.18  GB /  17.18  GB    512   B +  0 B   nvme3

```

igaw, 2025-04-22:
The kernel exposes single path devices differently compared to multipath devices.  For multipath devices you need to iterate over the subsystems into the namespaces. See how nvme-cli handles this:  `stdout_subsystem_topology` and `stdout_subsystem_topology_multipath`

closed 2025-04-22 by igaw