linux-nvme/libnvme #1013: nvme_scan_topology documentation and return value don't match

smithkyl, 2025-05-20:
Looking at src/tree.h, nvme_scan_topology's return value is described as the "Number of elements scanned". However, in src/tree.c, the function only ever returns 0 or less than 0 in the case of a scandir failure.

Not sure which one folks would like to fix, but it seems like either the function should return ctrls.num + subsys.num or the documentation should be updated to reflect the return value appropriately.

As a side note, if we are following the documentation strictly, returning 0 when no root is given makes sense, but it does seem more useful to return something like -1 and set errno to EINVAL for that scenario.

igaw, 2025-06-26:
Thanks for the report.

smithkyl, 2025-06-26:
Thanks for the fix!