linux-nvme/libnvme #1053: TLS key generation not comptiable with OpenSSL 3.0.x LTS.

igaw, 2025-08-22:
https://github.com/linux-nvme/libnvme/pull/1051 updated the PSK and it turns out that there is a compatibilty issue with OpenSSL 3.0.x LTS. With newer version of OpenSSL the test passes.

```
$ ./scripts/build.sh fallback
$ meson test -C .build-ci psk  -v
ninja: Entering directory `/home/wagi/work/libnvme-upstream/.build-ci'
ninja: no work to do.
1/1 libnvme / psk RUNNING
>>> MALLOC_PERTURB_=198 MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 LD_LIBRARY_PATH=/home/wagi/work/libnvme-upstream/.build-ci/subprojects/openssl-3.0.8:/home/wagi/work/libnvme-upstream/.build-ci/src:/home/wagi/work/libnvme-upstream/.build-ci/subprojects/json-c-0.18 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 MESON_TEST_ITERATION=1 /home/wagi/work/libnvme-upstream/.build-ci/test/test-psk
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
test nvme_export_tls_key hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_export_tls_key hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_import_tls_key hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_import_tls_key hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_export_tls_key_versioned hmac 0 NVMeTLSkey-1:00:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_export_tls_key_versioned hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_export_tls_key_versioned hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_import_tls_key_versioned hmac 0 NVMeTLSkey-1:00:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_import_tls_key_versioned hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_import_tls_key_versioned hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_generate_tls_key_identity host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 1 NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys iSbjiStwJ/1TrTvDlt2fjFmzvsytOJelidNnA+X5lEU=
ERROR: got 'NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys mJUDthe4jhFVFSnaBaydV/EHJK6OvIuw8xap5IkTnG0=', expected 'NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys iSbjiStwJ/1TrTvDlt2fjFmzvsytOJelidNnA+X5lEU='
test nvme_generate_tls_key_identity host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 2 NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys QhW2+Rp6RzHlNtCslyRxMnwJ11tKKhz8JCAQpQ+XUD8f9td1VeH5h53yz2wKJG1a
ERROR: got 'NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys J6B5sIVRCNLtZutDfmNnfPeqOFbnewwc8KEkhcOcO0dAWfdJYe/DrMyIC7znu00M', expected 'NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys QhW2+Rp6RzHlNtCslyRxMnwJ11tKKhz8JCAQpQ+XUD8f9td1VeH5h53yz2wKJG1a'
test nvme_generate_tls_key_identity_compat host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 1 NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys 66GuqV08TsAGII39teWUfwQwizjv06Jy8jOcX3NAAzM=
ERROR: got 'NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys mJUDthe4jhFVFSnaBaydV/EHJK6OvIuw8xap5IkTnG0=', expected 'NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys 66GuqV08TsAGII39teWUfwQwizjv06Jy8jOcX3NAAzM='
test nvme_generate_tls_key_identity_compat host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 2 NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys RsKmYJ3nAn1ApjjMloJFbAkLPivONDAX/xW327YBUsn2eGShXSjCZvBaOxscLqmz
ERROR: got 'NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys J6B5sIVRCNLtZutDfmNnfPeqOFbnewwc8KEkhcOcO0dAWfdJYe/DrMyIC7znu00M', expected 'NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys RsKmYJ3nAn1ApjjMloJFbAkLPivONDAX/xW327YBUsn2eGShXSjCZvBaOxscLqmz'
――――――――――――――――――――――――――――――――――――――――――――――――――――――――

```

The question is do we update the minimum requirement for OpenSSL (which is the min version) and or make the current code also work with OpenSSL 3.0.x.

cleech, 2025-08-25:
It looks like in older OpenSSL (prior to 3.3.1?) EVP_PKEY_CTX_add1_hkdf_info() acted as a "set1" function and you can't build up the info vector in multiple steps like I did.  Bummer.  I'll try and work up a fix.

cleech, 2025-08-26:
> can't build up the info vector in multiple steps like I did

Rather, this isn't a regression with the new changes but a long standing issue exposed by new tests.  It's a shame that openssl seems to be difficult to update as a meson wrapped dependency.

igaw, 2025-08-27:
>  It's a shame that openssl seems to be difficult to update as a meson wrapped dependency.

Indeed, the whole OpenSSL situation seems a bit tricky. Not even the PR to update to a more recent 3.0.x release is simple...

igaw, 2025-09-01:
FYI, after eff0ffef0273 ("linux: fix HKDF TLS key derivation back to OpenSSL 3.0.8") the master branch is still failing.
Haven't look into the details yet.

```bash
$ meson test -C .build-ci psk  -v
ninja: Entering directory `/home/wagi/work/libnvme-upstream/.build-ci'
ninja: no work to do.
1/1 libnvme / psk RUNNING
>>> MALLOC_PERTURB_=4 MESON_TEST_ITERATION=1 LD_LIBRARY_PATH=/home/wagi/work/libnvme-upstream/.build-ci/subprojects/json-c-0.18:/home/wagi/work/libnvme-upstream/.build-ci/src:/home/wagi/work/libnvme-upstream/.build-ci/subprojects/openssl-3.0.8 MSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 ASAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1 UBSAN_OPTIONS=halt_on_error=1:abort_on_error=1:print_summary=1:print_stacktrace=1 /home/wagi/work/libnvme-upstream/.build-ci/test/test-psk
――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――― ✀  ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
test nvme_export_tls_key hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_export_tls_key hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_import_tls_key hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_import_tls_key hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_export_tls_key_versioned hmac 0 NVMeTLSkey-1:00:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_export_tls_key_versioned hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_export_tls_key_versioned hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_import_tls_key_versioned hmac 0 NVMeTLSkey-1:00:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_import_tls_key_versioned hmac 1 NVMeTLSkey-1:01:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwNf9LrZ:
test nvme_import_tls_key_versioned hmac 2 NVMeTLSkey-1:02:VRLbtnN9AQb2WXW3c9+wEf/DRLz0QuLdbYvEhwtdWwP/w0S89ELi3W2LxIcLXVsDn8kXZQ==:
test nvme_generate_tls_key_identity host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 1 NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys iSbjiStwJ/1TrTvDlt2fjFmzvsytOJelidNnA+X5lEU=
test nvme_generate_tls_key_identity host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 2 NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys QhW2+Rp6RzHlNtCslyRxMnwJ11tKKhz8JCAQpQ+XUD8f9td1VeH5h53yz2wKJG1a
test nvme_generate_tls_key_identity_compat host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 1 NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys 66GuqV08TsAGII39teWUfwQwizjv06Jy8jOcX3NAAzM=
ERROR: got 'NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys mJUDthe4jhFVFSnaBaydV/EHJK6OvIuw8xap5IkTnG0=', expected 'NVMe1R01 nqn.psk-test-host nqn.psk-test-subsys 66GuqV08TsAGII39teWUfwQwizjv06Jy8jOcX3NAAzM='
test nvme_generate_tls_key_identity_compat host nqn.psk-test-host subsys nqn.psk-test-subsys hmac 2 NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys RsKmYJ3nAn1ApjjMloJFbAkLPivONDAX/xW327YBUsn2eGShXSjCZvBaOxscLqmz
ERROR: got 'NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys J6B5sIVRCNLtZutDfmNnfPeqOFbnewwc8KEkhcOcO0dAWfdJYe/DrMyIC7znu00M', expected 'NVMe1R02 nqn.psk-test-host nqn.psk-test-subsys RsKmYJ3nAn1ApjjMloJFbAkLPivONDAX/xW327YBUsn2eGShXSjCZvBaOxscLqmz'
―――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
1/1 libnvme / psk FAIL            0.01s   exit status 1

```

closed 2025-09-04 by hreinecke

igaw, 2025-09-04:
I did not merge #1059, you just closed it. master is still failing

PR #1073 (2025-09-17): linux: use EVP_PKEY_CTX_add1_hkdf_info only once in compat function
linux: use EVP_PKEY_CTX_add1_hkdf_info only once in compat function

OpenSSL prior to 3.3.1 had an issue with EVP_PKEY_CTX_add1_hkdf_info()
where it acted like a 'set1' function instead of an 'add1' as
documented.  Work around that by building the entire info vector outside
of the OpenSSL API and only calling this function once.

This is the same workaround used in commit eff0ffef0273 ("linux: fix
HKDF TLS key derivation back to OpenSSL 3.0.8").

Fixes: #1053